edutec/Snap4Arduino-old-huge #49: Name installers and tar files with platform and version

fhunleth, 2015-01-05:
The problem is that the installers and tar files either have the same or very similar names (Snap4Arduino.tar.gz or Snap4Arduino.zip) and people sometimes grab the wrong one and don't know it. I renamed the files manually when I copied them locally for our use, but would it be possible to add the platform and version number to the names on GitHub and the S4A site. E.g. Snap4Arduino-1.0-beta-win32.zip, Snap4Arduino-1.0-beta-osx.zip, Snap4Arduino-1.0-beta-linux-x64.tar.gz, etc.. It would also be nice to have more granularity in the version strings since it looks like 1.0-beta has been 1.0-beta through a few changes, but the different filenames per platform is the big thing that would help me.

I don't mind sending the pull request if you're ok with this, but let me know your preferences on naming.

bromagosa, 2015-01-06:
Sure thing, please go ahead and issue a pull request!

My only preference on naming is that we avoid using *Linux* to refer to the whole system. I know I'm weird and picky, but I think either *GNU/Linux* or just *GNU* are more accurate. To make my point stronger, imagine we called the *MacOSX* package *Snap4Arduino-XNU*, instead of *OSX*... A package for Linux, IMHO, should be something strictly dedicated to the kernel, not the whole system.

You're completely right on granularity. I've just been lazy... every time I update the version name I have to change it in so many places, including the official site, that I'm inclined to just keep the same number when differences are not too noticeable. But yeah, let's be more consistent from now on.

What about naming this one 1.0.1 beta?

edutec, 2016-02-02:
This issue was moved to edutec/Snap4Arduino#8

closed 2016-02-02 by edutec