edutec/Snap4Arduino-old-huge #50: servo and PWM not working (no pin choice)

bschwand, 2015-02-06:
It seems there is a bug with the set servo and set PWM blocks.  (or I do not know how to use them....) In that case I apologize.

I installed the very latest firmata 2.4.0 release on an arduino nano 328p from the firmata website https://github.com/firmata/arduino
I tested that a digital pin set as digital ouput works, so I know communication functions.

Then, I used the block "set digital output to ..." in order to define the type of the output (for example to servo )
Then I placed a block "set servo xxx to clockwise. 

the problem is that the servo number field is always blank, no number can be chosen. Same thing with "set PWM pin xxx to".
I noticed that those fields for digital IO setup are blank until the arduino is connected, so I assume upon connecting the arduino, the proper pin numbers are read from firmata, but somehow no servo or PWM is found ? although the capability is definitely there...
I also tried the standard firmata version included with Arduino IDE, same results.

I also installed PyMata ( a python binding to firmata) and also tried FirmataPlus on the arduino which comes with PyMata. From Python, I verified PWM and servo access works on my arduino board. So I think it is an issue with Snap4Arduino.

I will be happy to provide any additional information needed, and do tests. 
I can try to debug, could someone indicate the entrypoints for the Arduino stuff,. i.e. in Snap4Arduino, which code is executed when one clicks on the "Connect to Arduino" Button ?

bschwand, 2015-02-07:
I tried debugging by enabling the toolbar in package.json, but snap4arduino crashes when I try to step into the Arduino.firmata.Board() function. How can I debug the instanciation ? I believe something is handled wrong at connection since the PWM and servo pins are not retrieved correctly.

bromagosa, 2015-02-09:
Hi @bschwand. I'm afraid I don't understand your issue...

If I get it right, you're mistaking what the set digital output block does. This block is meant to take a boolan as an argument, for instance:

[image]

If what you want is to configure a pin, you should use the pin setup block:

[image]

After running this block, the servo pins will be updated and you'll be able to send them commands:

[image]

Same goes for PWM:

[image]

Of course no PWM or servo pins are found when you connect the Arduino, the software cannot guess what you want to do with your pins unless you tell it ;)

closed 2015-02-09 by bromagosa

bschwand, 2015-02-09:
Hello,

thanks for the reply !

No actually, I did not express myself well. So te be more precise:

- I used the pin setup block to configure the pin to servo mode
- then I placed my "set servo xx to xxxx"

at that point, the set servo block still offers no choice for the servo
number. But thinking what you said "After running this block" I tried to
run the setup block, after that a choice appeared in the "set servo xx to
xxx" block. So I can use pwm and servo (well not physically tested I need
to wire a servo... but it seem to accept the command). That is great, thank
you !

So I understand now that it is only after configuring AND running the
configuration block, that the pins appear available for PWM and servo.
However, you will admit this is a little counterintuitive, considering that
the digital and analog blocks, _do_ show the correct pin choices
immediately after connecting the Arduino.

I have been looking at all the Firmata implementation on the arduino, as
well as the Firmata protocol and also the Firmata javascript bindings. It
is clear that after connection, the Firmata capability query which tells
how many digital and analog pins there are, also provides the modes allowed
for each pin. (I checked also with PyMata, a nice python Firmata Client to
dump the capability query results, and the pin modes are also there for pwm
and servo).

The "set servo xx to xxx" and "set PWM pin xx to xxx" block could well be
updated and show the correct pins, right after the connection. I think they
should, it would be a lot more logical, wouldn't it ?

Also, not all pins support PWM or servo mode, the capability query will
return correctly which ones do. Using this will avoid errors, where one
would set a certain pin to a mode which is not supported and then try to
use it. Specially for something that is destined for education and use by
kids, I think it would be good to look in to this.

I unfortunately have not been able to dig deep enough because somehow the
node debugger crashes whenever I try to debug firmata and serialport
modules, in order to find out where exactly this happens. I did notice that
at the Snap block level, the arduino "board" object, does not have all the
pin modes information, so I would think probably the js Firmata binding is
not processing the capability query correctly, not sure. If you have any
tip on debugging snap4arduino all the way down to serialport, I would very
appreciate them, thanks.

Conclusion: it seems to work, but I think the initialization of the servo
and PWM block need to be looked into and fixed to be more "Child-friendly"
 :-)

Bruno