edutec/Snap4Arduino-old-huge #56: Uncaught node.js Error when download extra costumes

nuzza412, 2015-04-29:
Hi there,

I would like to report a bug in the software. Whenever I try to download some extra costumes this "Uncaught node.js Error" comes out. Please see the screenshot for further information.
Many thanks
[image]

bromagosa, 2015-04-29:
Hi @nuzza412, thanks for the report!

How are you trying to import these costumes? Drag and drop?

nuzza412, 2015-04-29:
Hi @bromagosa I tried to use the drop down menu. Please see the attachment.
[image]

bromagosa, 2015-04-29:
Ouch, that's right... I'll take a look as soon as I can.

Thanks!

nuzza412, 2015-04-29:
No prob! Thank you!

nuzza412, 2015-04-29:
The same error is displayed if you try to download the sounds :(

commit d216a7e2 (2015-05-14): fixes #56
closed 2015-05-14 by bromagosa