edutec/Snap4Arduino-old-huge #65: Drop downs in blocks not available

mryantho, 2015-10-14:
The drop down selectors in the variable blocks, and the Arduino blocks do not work.  Trying to use the dropdown selector to choose a variable has the behavior of moving it.

Have tried re-downloading and re-installing.

Win 7
Snap 1.0.5-beta

coalman321, 2015-11-22:
try doing all the board stuff first. this includes loading the fimata and connecting the board under the arduino menu then hit connect and select the com port. after this most dropdowns become active.
this worked for me

bromagosa, 2015-11-23:
Yes. The rationale behind this is that Snap4Arudino cannot know which board are you going to connect to your computer until... well, until you connect it ;)

So it's impossible to know the pinout of a board that's not yet connected, and since we're supporting all kinds of Arduino boards, we can't provide a default configuration that will work across different models.

closed 2015-11-23 by bromagosa

mryantho, 2015-11-24:
Yes that does make sense, and I was having trouble connecting an Arduino at
times on my computer.  There seems to be some other issue with using so
many Arduinos on different computers in my lab where some UNOs work with
some PCs but not others.  I think it has something to do with USB/Driver
issue.  It has been frustrating since students don't have admin access to
get to Device Manager.

On this issue, the culprit was that SMART Board's screen annotating
"floating tools" was somehow messing up the clicking of the dropdown.