edutec/Snap4Arduino-old-huge #69: Problems reading digital input

elaval, 2015-11-26:
I can not read digital inputs (always get a false value) in the latest version (1.0.6) but it does work on version 1.05 (8c12f4b5fe53be13c2f8f3b4503c3ddd5d044dda).

I am testing on a Mac (would appreciate if you could test if this works for you to check if it is a local issue with my configuration or something in the code).  Maybe the new version of serialport has some differences for reading & updating the values in digital inputs.

bromagosa, 2015-11-27:
I'll check it right away, thanks for reporting!

p.s. I know I owe you a review... we've been super busy with the annual Citilab conference, but we have tested the kit and overall we like it! I'll write back to you as soon as I can! :)

elaval, 2015-11-29:
After your conference hype I would love to chat abt your board review.  In the meantime let me share what I have explored in relation to the digital read issue that I found (at least in the Mac version).

In Snap4Arduino we get the digital value of the respective pin using the ´value´attribute in the board.pins[pinNumber] object.
s4aThreads.js:
```javascript
Process.prototype.reportDigitalReading = function (pin) {
	var sprite = this.homeContext.receiver;

	if (sprite.arduino.isBoardReady()) {
		var board = sprite.arduino.board; 

		if (board.pins[pin].mode != board.MODES.INPUT) {
			board.pinMode(pin, board.MODES.INPUT);
		}
		return board.pins[pin].value == 1;
	} else {
		throw new Error(localize("Arduino not connected"));		
	}

}
```
It seems that in previous firmata.js versions (i.e. 0.3.3) the ´pin´ attribute was automatically updated even if we did´t call the asynchronous API function digitalRead:
```javascript
digitalRead(pin, callback) 
```
But it seems that in more recent versions of firmata (0.3.8 onwards) the pin attribute is not automatically updated.  
Try the following node sample code:

```javascript
// index.js
var firmata = require("firmata");
var board = new firmata.Board("/dev/cu.usbmodemFA131", function() {
  console.log("connected");
  board.pinMode(2,board.MODES.INPUT);

  setInterval(function() {
	console.log(board.pins[2].value);
  }, 100)

});
```
It is expected to print the digital value of pin[2] every 100 ms.  
If we use firmata 0.3.3 - npm install firmata@0.3.3 - we would get the following output  when we change the input signal on port 2:
```
$ node index.js 
connected
0
0
0
0
1
1
0
0
1
0
0
1
1
```
If we use firmata 0.3.8 - npm install firmata@0.3.8 - we would get the following output when we change the input signal on port 2:
```
$ node index.js 
connected
0
0
0
0
0
0
0
0
0
0
0
0
0
```
THE PIN ATTRIBUTE IS NEVER UPDATED

A possible solution is to add - once - a digitalRead instruction for the Input Pins.  With this it seems that the pin value starts to be updated:
 
Check this code which introduces the board.digitalRead() instruction:
```javascript
var firmata = require("firmata");
var board = new firmata.Board("/dev/cu.usbmodemFA131", function() {
  console.log("connected");
  board.pinMode(2,board.MODES.INPUT);

  board.digitalRead(2,function(value) { 
  	// no instruction required
  })

  setInterval(function() {
	console.log(board.pins[2].value);
  }, 100)

}); 
```
And the output with firmata 0.3.8 becomes:
```
$ node index.js 
connected
0
0
0
0
1
1
0
0
1
0
0
1
1
```

Not sure how much changes we need in the Snap4Arduino code to implement a fix (not even sure if the problem is presented in other platforms besides OSX where I have tested it).  Not sure what happens if we start a board.digitalRead() , and then we change the pin mode.  Much to be explored before a final solution.  

I would appreciate if you could report if you have been able to replicate the problem.

closed 2015-11-30 by bromagosa

bromagosa, 2015-11-30:
I've tested it extensively, and calling ``digitalRead`` seems to fix the problem with no side effects even if you later reconfigure that pin.

Thanks for finding this! The new version is already online at [s4a.cat/snap](http://s4a.cat/snap)

commit ba3a37e6 (2015-11-30): fixes #69